analogdevicesinc/scopy #174: Setup option on main menu

msuciu, 2017-07-12:
I thing it will be a good idea to add a new option on the main menu called "Settings" on which the users can chose how they want to use the application. 
This menu can have two or more options, for example "Windows mode" and "Android mode".
In this way there are no limitations regarded on operation systems, touchscreens or mouses, etc and you can test how this application will work on Android in a Windows environment.
So, if you chose the "Windows mode" option you can use the application like any other Windows application, like left and right menus or options, scrolling, etc.
Or, if you chose the "Android mode" option, since you don't have the right click option, or a mouse, on a smartphone or a tablet you can implement a pie menu or a radial menu for more options.   
[image][image]

rgetz, 2017-07-12:
I disagree - this is the opposite of what we want to do (I think).

We one one GUI, which people don't need to re-learn based on the platform they are using; or get frustrated since they know how to do something on OS, but not on the other.

msuciu, 2017-07-12:
In this way all features which use right click are not allowed, like zoom out and to do this is necessary to use the instrument menu, or other left click tricks, so the application is used in a slower way. This is was only a suggestion to use the application faster.

pcercuei, 2017-07-13:
We can handle right-clicks as shortcuts for actions that can be done a different way.

msuciu, 2017-07-18:
What if we develop an option for stroke commands, like drawing a symbol on the screen to quick call an option? This should work on any OS or device. This could be accomplish by adding a check box to enable stroke commands. This check box can be enabled by the user and then to be automatically disabled after a valid stroke command is executed.
[image]

mbanciso, 2017-07-18:
In my experience this only complicates things and makes them trickier to use

pcercuei, 2017-07-18:
Agreed with @mbanciso. The UI should be a no-brainer to use.

rgetz, 2017-07-18:
UI should be obvious and intuitive. 

Instruments do not have right clicks, or obtuse gestures to use them. We should not either. 

https://youtu.be/JDKtllS-Lw0

msuciu, 2017-07-18:
Close this issue since thinks are now clear.

closed 2017-07-18 by msuciu