analogdevicesinc/scopy #176: Amplitude meaning miscorrelation between Scopy and Activity Lab Material

amiclaus, 2017-07-13:
The Amplitude in Scopy represents the peak-peak value.
Going through the Lab Material for m2k, I've encountered the following statement: 'Use the first waveform generator as source Vin to provide a 1V amplitude (2 Vp-p)' (Source: [Activity 1. Simple Op Amps](https://wiki.analog.com/university/courses/electronics/electronics-lab-1))

Therefore, I think that the amplitude value should have the same meaning both in Scopy and the Activity Lab Material.

(A similar tool to Scopy refers to the amplitude value as peak value).

pcercuei, 2017-07-13:
I guess the Lab Material could just say "Use the first waveform generator as source Vin to provide a 2V amplitude".

msuciu, 2017-07-13:
I thing the Amplitude of the waveform in the Signal Generator should be limited to 5 since you don't change the peak-to-peak value.

pcercuei, 2017-07-13:
The amplitude in the Signal Generator is the peak-to-peak value, so the maximum is effectively 10V. I don't understand what you mean.

msuciu, 2017-07-13:
You're right, but still you need to limit the Amplitude of the waveform in the Signal Generator.
[image]
[image]
[image]
[image]

pcercuei, 2017-07-13:
No, you don't. Generating 10V should give you a non-trimmed sinewave as long as the calibration is done properly.

rgetz, 2017-07-13:
Just because Waveforms was describing it wrong (amplitude as 1/2 of peak-to-peak), doesn't mean that scopy should do the same.

We need to update the lab materials.

amiclaus, 2017-07-13:
OK, then the lab material will be updated. Since a decision was taken on this topic, I will close the post.

closed 2017-07-13 by amiclaus

mthoren-adi, 2019-01-27:
This was noticed again by Zeke - there is still some ambiguity. Keysight and Rigol manuals both define amplitude as the peak-to-peak value, but the units are always very explicit. From Keysight 33201A manual:
[image]
From Rigol DG1022:
[image]

Sedra and Smith and Art of Electronics both define the amplitude of a sinewave as the peak value measured from zero - the "A" in the formula V=A*sin(2*pi*f*t).

Would it make sense to change the unit labels to "Volts p-p", "mVolts p-p", "uVolts p-p" in the waveform generator, as well as oscilloscope measurement display?

zrandolph, 2019-01-28:
Merriam-Webster defines amplitude as:

> Definition of amplitude. 
> 1 : the extent or range of a quality, property, process, or phenomenon: such as. a : the extent of a vibratory movement (as of a pendulum) measured from the mean position to an extreme. b : the maximum departure of the value of an alternating current or wave from the average value.

This is the definition that I was taught all throughout school; it will be frustrating and confusing for students to discover our own decision of what the definition should be, and attempt to do schoolwork juggling the two definitions.

The unfortunate thing about the official definition is that, for asymmetric waveforms, "amplitude" loses usefulness because there are two different meaningful extents from the mean.

I support Mark's idea of specifying peak-to-peak instead, as Keysight and Rigol do. They must have encountered the same issue.

damercer, 2019-08-21:
All of academia uses the definition of amplitude as the maximum deviation from the average. Regardless of what bench instrument manufacturers do or have done in the past, ADI should follow academia on the use of Amplitude. This will make it easier to use out stuff in their classes and labs. End of discussion!

Doug