analogdevicesinc/scopy #177: Volts/Div reset to 1 after zoom out

msuciu, 2017-07-13:
Connect channel 1 of the Signal Generator on the channel 1 of the Oscilloscope. Generate any signal on channel 1. Go to the Oscilloscope and change the Volts/Div value to any value greater than 1, zoom in and zoom out. The Volts/Div value looks like is reset to 1. A similar thing is happening when the Volts/Div value is changed to any value lower than 1, but the measured signal is limited in amplitude.

Original signal:
[image]

After zoom out when Volts/Div value is greater than 1:
[image]

After zoom out when Volts/Div value is lower than 1:
[image]

DanielGuramulta, 2017-09-06:
Fixed with commit:https://github.com/analogdevicesinc/scopy/commit/8159977b625de12d0a54aaaa1f1af8704cec47bf

amiclaus, 2017-09-08:
On Scopy build: 152efe0(v0.7) the bug is fixed.

closed 2017-09-08 by amiclaus