analogdevicesinc/scopy #178: Signal Generator interruptions when Oscilloscope is started

msuciu, 2017-07-13:
Generate a signal using the Signal Generator and visualize the generated signal with an external oscilloscope. While the Signal Generator is running start the Scopy Oscilloscope.

[image]
[image]

damercer, 2017-07-14:
Glitch seems to be small and probably a non-issue when user is only using M2K to do their experiments / lab measurements. Can sort of expect short interruptions when the software makes changes to what the board is doing.

Doug

pcercuei, 2017-07-17:
That's not a software issue, the signal generator just pushes a cyclic IIO buffer then idles. Probably a DMA issue then.

lclausen-adi, 2017-07-24:
@msuciu can you share the pattern generator and scope configuration under which this issue was observed.

msuciu, 2017-07-24:
Sure. Keep running the Scopy Signal Generator and connect an external oscilloscope on one of the channels. If you are using build 3aae9de use channel 1 because channels 2 is not working for some reason. Then press the play button on the Scopy Oscilloscope. When you do this observe on the external oscilloscope the issue described above. The configuration looks like is not important. I hope this helps. 
[image]
[image]

adisuciu, 2018-01-12:
https://github.com/analogdevicesinc/scopy/issues/316 addresses this issue.

closed 2018-01-12 by adisuciu