analogdevicesinc/scopy #180: Signal Generator: Waveform's amplitude is not stable on lower frequency

drbautista, 2017-07-14:
OS: Windows 7 64bit
Scopy Version:  b210916
Firmware Version: v0.14

      1.	Connect M2K to scopy.
      2.	Connect scope ch1+ to AWG ch1 and scope ch- to gnd.
      3.	Open signal generator instrument and set it to the following setting:
                  a.	Amplitude: 6V
                  b.   Frequency: 1kHz
                  c.	Offset: 0V
                  d.   Phase: 0degrees
      4.	Run signal generator and oscilloscope
      5.	The parameter reading in the oscilloscope is correct here.
                    
[image]

      6.	Change the frequency to 100Hz
      7.	The parameter reading in the oscilloscope will be wrong.

[image]

This behavior is present on the different waveforms. 
I tried looking at these waveforms on an external oscilloscope to make sure that the one with problem is the signal generator.
There are frequencies between 100Hz to 999Hz showing unstable amplitude.

damercer, 2017-07-14:
I added a comment similar to this to issue #170 

Doug

pcercuei, 2017-07-17:
I'll close this bug as it's a dup of #170. We can continue the discussion there.

closed 2017-07-17 by pcercuei