analogdevicesinc/scopy #181: Signal Generator: Constant voltage can't generate ±5V

drbautista, 2017-07-14:
OS: Windows 7 64bit
Scopy Version:  b210916 
Firmware Version: v0.14

Connect AWG ch1 to scope ch1+ and scope ch1- to gnd.
Connect AWG ch2 to scope ch2+ and scope ch2- to gnd.
Open Signal generator, set both channels to constant waveform and set channel 1 or 2 to +5V or -5V.

Run signal generator and voltmeter.

[image]

The signal generator won't be able to generate +5V or -5V voltage.
Sometimes it may reach up to 4.5V to 4.7V. It's unstable in the configuration from 4V to 5V configuration or -4V  to -5V configuration.

pcercuei, 2017-07-17:
This is most likely a consequence of #170.

dNechita, 2017-07-24:
Indeed, the bug is a consequence of issue 170 which has been fixed with commit https://github.com/analogdevicesinc/scopy/commit/34b66263e6f7439c4a9f100651b1c4635c3f0ea7.

amiclaus, 2017-07-25:
On Scopy build: f759590 (v0.6) the bug is fixed.

closed 2017-07-25 by amiclaus