analogdevicesinc/scopy #183: Horizontal cursor lines are hidden by the plot

msuciu, 2017-07-14:
On the oscilloscope when you enable cursors and you move channel 2 vertical position outside the plot the vertical cursor lines are hidden by the plot. They are comeback in front of the plot only if you move the horizontal position.
[image]
[image]

msuciu, 2017-07-14:
To be more specific, this bug is "activated" when Volts/Div value for channel 2 is changed.

msuciu, 2017-07-14:
Also, if you increase the time base at maximum the vertical cursor line length is changed depending on channel 2 vertical position.
[image]
[image]

rgetz, 2017-07-15:
Just to be clear - the measurement/cursor lines should be "behind" the waveform plots. (The cursors shouldn't obscure the voltage lines from either channel) - but they shouldn't dissapear either. 

They should move with the voltage scale. For example -  A voltage cursor on channel 1, if you change the channel 1 voltage scale - the positions of the cursors on the screen will change (and may even move off the screen, to the top or bottom), but should keep the same voltage position. 

Measuring on the time axis should do the same. It should track the seconds offset, not the relative position on the screen; and be between the grid (on the bottom), and the voltage plot (on the top).

mbanciso, 2017-07-17:
I looked at 2 brands of bench scopes and the cursors are fixed to the screen. 
They don't change with time base or v/div

dNechita, 2017-07-17:
Fixed in commit https://github.com/analogdevicesinc/scopy/commit/3eb261041d9d863203de5af4adc42bde6cb769c1

msuciu, 2017-07-18:
I confirm this fix after testing with Scopy 3eb2610.

amiclaus, 2017-07-24:
On Scopy build: 3aae9de (v0.6) the bug is fixed.

closed 2017-07-24 by amiclaus