analogdevicesinc/scopy #184: Switching between channels menu interrupt the generated signals

msuciu, 2017-07-14:
When you switch between channels menu the generated signals are interrupted. It is normal when you change the configuration for any channel, but it's not ok when the configuration is the same.

[image]

rgetz, 2017-07-14:
Does it make a difference on the sample rate of the generated signal?

It looks like that is a super slow signal...

msuciu, 2017-07-17:
Yes, depending on the sample rate this interruptions take a different amount of time. Below you can see how much time takes this interruptions for a generated signal of 1 Hz, 1kHz and 1 MHz when I switch between channels menu.
[image]
[image]
[image]

msuciu, 2017-07-18:
I confirm this fix after testing with Scopy 3eb2610.

amiclaus, 2017-07-25:
On Scopy build: f759590 (v0.6) the bug is fixed.

closed 2017-07-25 by amiclaus