analogdevicesinc/scopy #188: Signal Generator: Channel 2's color indicator for configuration menu is similar with channel 1's

drbautista, 2017-07-18:
OS: Windows 7 64bit
 Scopy Version:  3eb2610
 Firmware Version: v0.14

The color indicator for channel 2's configuration is similar with channel 1 which is orange.
I think it would be good if the color is similar to the waveform and the on/off button for channel 2 to avoid confusion.

[image]

dNechita, 2017-07-18:
This is a duplicate of #10

pcercuei, 2017-07-18:
The tabs are underlined with orange in the theme, independently of the colors of the channels. Maybe we should change that to the same blue we use everywhere.

DanielGuramulta, 2017-08-25:
Fixed with commit: 6fd7ece

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus