analogdevicesinc/scopy #189: New option for signal senerator

msuciu, 2017-07-18:
It is possible to have a new option on the signal generator which will allow to not synchronize the generated signals, like each channels to be independent?

mbanciso, 2017-07-18:
That's why the phase controls are there

msuciu, 2017-07-18:
If you want to keep a generated signal synchronized with an external one and change the configuration for the second generated signal is not possible in this configuration.

mbanciso, 2017-07-18:
we don't have triggers for the outputs yet. only in stage 2

damercer, 2017-07-18:
Since both signal generator channels use the same sample clock (i.e. time base ) they are by definition synchronized. If say one is at 1 KHz and the other is exactly twice that at 2 KHz the only thing that can be different is the phase or delay. Or if one was 1 KHz and the other was 1.001 KHz they would appear to be not locked to each other but in fact are every 1 second ( the 1 Hz difference ).

If you do implement external triggering for one channel and not the other then there will have to be a jump in phase between the two ( a discontinuity ) every time a trigger comes along.

Doug

msuciu, 2017-07-18:
The idea was to change the configuration for one channel without to disturb the other one. As I understand if a channel configuration is changed the other channels is also affected by that to keep both channels is phase. Anyway, probably this feature is not so important or useful.

rgetz, 2017-07-18:
I will always go back - when is this necessary for a student lab? If it is not - please close this as nack'ed.

We need to focus on closing bugs, not adding new things that are not important to the key objective - student learning.

closed 2017-07-18 by msuciu