analogdevicesinc/scopy #195: State saved after using Network analyzer?

damercer, 2017-07-22:
This question relates to what appears to be a bug in what gets saved in the start-up configuration file after using the Network analyzer. Now after using the network analyzer AWG channel 2 seems to be permanently disabled. With the Network analyzer off and the signal generator open and running, both channels set and enabled, nothing is coming out of the output 2 as seen with the scope. Channel 1 work fine. Condition persists after unplugging the M2K and restarting Scopy. Also on a different M2K.

I seem to remember a configuration file being saved somewhere but can't seem to find it. I looked in AppData/Roaming/Analog Devices but don't see a Scopy directory?

Anyway, something has messed up the configuration and I can't get it back.

Thanks

Doug

rgetz, 2017-07-22:
There should be a file (a Scopy.ini) that you can move to the trash can. I don't remember the exact name or location - I will test it out - and update this later.

damercer, 2017-07-22:
More info: If I choose channel 2 as the reference channel in the network analyzer it generates an output but still will not generate an output when I go back and run signal generator ( and scope ).

Doug

damercer, 2017-07-22:
I found the Scopy.ini file. It is in the ADI directory under Roaming. Strange that I didn't see that directory the first time? Anyway deleting the Scopy.ini did not fix the problem. Still no output on channel 2.

Doug

damercer, 2017-07-22:
Un-installing Scopy, Re-installing Scopy, Rebooting computer did not fix issue either.
However everything works fine on the other computer ( ADI laptop ) running Scopy ea15025.
Have not gotten around to loading the latest build on that one.

Doug

rgetz, 2017-07-22:
Closing Scopy, deleting the ini file, restarting should be enough to reset the state of everything (that I know). 

Maybe Dan will need to comment.

damercer, 2017-07-22:
OK, loading older version ( same ea15025 ) fixed the problem. So there must be something wrong with Scopy version 3aae9de. Probable actually unrelated to the Network analyzer. I just happened to notice that Channel 2 was missing after playing with the network analyzer.

Doug

dNechita, 2017-07-24:
There was a bug in the signal generator tool. Fixed with https://github.com/analogdevicesinc/scopy/commit/928a4d90a5c31893d33c17f04eefa84bbdf5fd64

amiclaus, 2017-07-25:
On Scopy build: f759590 (v0.6) the bug is fixed.

closed 2017-07-25 by amiclaus