analogdevicesinc/scopy #196: Cursors value should be corelated with plot divisions and subdivisions

msuciu, 2017-07-24:
When you place a cursor exactly on a division or on a subdivision on the oscilloscope plot the cursor value should be, in my opinion, a reference value. For example, if you set 1 Volt/division and you place a vertical cursor on the first division above 0 the cursor value should be 1 Volt.
[image]

damercer, 2017-07-24:
Another way to ask this is do we want the cursors ( and or markers ) to snap to the grid? Then the question is how wide is the snap window, i.e. how far can you be from a grid line ( placed between grids ) before it snaps. Doing this wrong can be very frustrating / annoying to the user.

On the other hand sometimes you might want to in effect snap to the signal ( like a peak or valley ). 

Doug

msuciu, 2017-07-24:
Probably some users will want this option and other not. So, a check box in GUI to enable or disable snap to grid will solve this problem. Another think which can be taken in consideration is to add in GUI an input box to enter the cursor position.

rgetz, 2017-07-24:
I don't know of any instrument that does this. (snap to grid? or waveform?)

Do you have an example?

msuciu, 2017-07-24:
http://wiki.lab-nation.com/index.php/Oscilloscope_functionality (for snap to grid)
Example bellow for cursor position.
[image]

rgetz, 2017-07-24:
I guess we need to be more pedantic on the terminology. 

"Snap to grid" :  as the cursor moves over the grid, it automatically “catches” every graticule. The cursor moves from minor point to minor point, with no way of moving between.

"Snap to waveform" : as the cursor moves over the grid, it automatically “catches” every peak or valley. The cursor must be more than 5 pixels away from a waveform, or it snaps to the peak, when scrolling over a peak, it jumps to it (when it's within 5 pixels), stays on it while the knob continues to be rotated, and then jumps 5 pixels away from it. I would think this would be super annoying most times, and super useful some times.

"Tolerance" : the reported tolerance of a reading is always going to be +/- 1 pixel, not a constant voltage tolerance - there isn't any better we can do. When calculating the cursor position, it's important to take this into account, and not create a mathematical representation of the answer. For example - if one pixel is ~10mV, when the cursor position calculates to 2mV, it's better to display 0V, rather than 2mV.

For reading tolerance - sure - I agree - the readings should make sense with the tolerance they have, and be rounded/truncated to things that make sense based on the graticule, and voltage settings.

snap - I'm not so sure of.

msuciu, 2017-07-24:
So snap to waveform could be something like this (if we want to add modes for cursors). Basically, a function which depends on the signal and plot (vertical or horizontal settings). And snap to grid will be only dependent on the plot (vertical or horizontal settings). [image]Thanks a lot for the explanations. Maybe if we implement a "graticule magnifier" around the cursor position or a label to see the graticule value the snap to grid will be useful especially if Scopy will be used with a smartphone where some simple operations can be performed without to open a menu. Just an idea ... :)

acozma, 2018-01-17:
Should consider the snap to grid and waveform features for phase 2. These features could prove to be useful especially for touchscreens.

Must have GUI options to enable/disable the snap.

adisuciu, 2019-07-01:
The track waveform functionality has been implemented. I don't see any use case for snap to grid. I'll close this.

closed 2019-07-01 by adisuciu