analogdevicesinc/scopy #200: Spectrum Analyzer issues

CElkhoury, 2017-07-24:
Few issues encountered:
1. The marker did not display the number anywhere. and couldn't add a second marker.
2. Peak Hold didn't seem to hold peaks at all times. 
3. The units vertical and horizontal don't need to be added to each tick mark only one ((MHz) below and one (dBFS) on the left hand side. Also the freq numbers don't need two decimal points. and the vertical numbers can be only 3 digits. (example  10.0, 20.0, 100, 110)
Charly

dNechita, 2017-07-25:
1. I'm currently working on improving the markers functionality.
2. Peak Hold will keep the peaks for the last N acquisitions. The N can be set with the 'Averaging' control. However, Peak Hold Continuous will hold the peaks from the moment you started the capture until you stop it and that probably what you wanted to use. If not, let me know what the problem was.
3. This is good feedback and should work when the scale is linear. But for the logarithmic scale I think we want to see on the frequency axis something like this: 1Hz, 10Hz, 100Hz, 1kHz, 10kHz, etc. instead of 0.001, 0.01, 0.1 1, 10, etc. (kHz being the unit for the axis in this example).

adisuciu, 2017-10-20:
These features are implemented in  cb7f5f9

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the features are implemented.

closed 2017-11-21 by amiclaus