analogdevicesinc/scopy #203: When will X vs Y plotting be implemented?

damercer, 2017-07-25:


dNechita, 2017-08-07:
I've added a working version of the X vs Y plot. https://github.com/analogdevicesinc/scopy/commit/01548dc6fa422625df308116d5da23d3474bc333
Currently it only shows Channel 1 vs Channel2 so it needs an improvement in order to have the option to select the channel source for each axis.

damercer, 2017-08-07:
> I've added a working version of the X vs Y plot. 01548dc

It will also need to allow Math channels on either axis as well. You 
would ideally need to have two different Math equations at the same time 
one for X and one for Y. So each axis could be CH1, CH1, Math1 or Math2. 
Just including this so it doesn't get overlooked in the process.

Doug

AlexandraTrifan, 2017-11-16:
Starting with commit c9850e3996eb5e03ef6f20ae67b26dd9cb7e325c the user can select which channel should be used for the X axis and for the Y axis.

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the enchancement is implemented.

closed 2017-11-21 by amiclaus