analogdevicesinc/scopy #204: Inappropriate display of the signal amplitude

amiclaus, 2017-07-27:
Scopy version: dd0a1a8

When going through MHz frequencies the amplitude of the the signal displayed on the plot does not correspond to the set amplitude.

[image]

adisuciu, 2017-08-01:
This issue is related to how the samples are generated at high frequencies. There is not much we can do to prevent this. We need to define limits for each waveform so that they are generated with minimal distortions.

rgetz, 2017-08-01:
I'm not sure what you mean.

if someone says 10Vp-p, then that is what should come out the pins. We don't spec how good the signal will look (distortions/harmonics/etc). Other equipment does the same. The only thing that is accurate is the p-p output value.

-Robin

adisuciu, 2017-10-20:
This was fixed with f3e3f436ce07e6fc18038d2370f5f0a8163951df and f2876e55e1d1ece4a1b2f3d358a623d9a0368e55. The preview will display the ideal waveform.

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the bug is fixed.

closed 2017-11-21 by amiclaus