analogdevicesinc/scopy #205: Disabling  a channel doesn't disable the selected measurement of that channel

tagoylo, 2017-07-27:
scopy build: [dd0a1a82](https://github.com/analogdevicesinc/scopy/commit/dd0a1a82cbb3bf56649dedb1fa144af51680c2d9)

Steps to reproduce:

Activate measurement period and frequency for both ch1 and ch2

Then disable channel1, measurements of channel 1 stays on the display.

Same behavior happens even if both channels are disabled and there is no trace displayed. The measurements remain until deleted. Disabling a channel doesn't disable the selected measurement of that channel.

[image]

dNechita, 2017-07-27:
This is a duplicate of https://github.com/analogdevicesinc/scopy/issues/91

As far as I'm concerned this isn't a problem. It doesn't do anything harmful. In fact it gives you a little bit of flexibility by hiding the waveform (in case things get crowded on the plot canvas) while keeping the measurements. The measurements can be easily disabled at any time from the 'Horizontal' or 'Vertical' drop-downs. Moreover, you could use the 'Delete All' button for the channel you are about to hide.

AlexandraTrifan, 2017-10-17:
This is not applicable for the latest version.

closed 2017-10-17 by AlexandraTrifan