analogdevicesinc/scopy #207: Plus button doesn't increase frequency until 37.5MHz, stuck at 20MHz

tagoylo, 2017-07-28:
Scopy build: [8406b304](https://github.com/analogdevicesinc/scopy/commit/8406b3043e72ebd5755094674c4a57aa839a565a)

Steps to reproduce:

Set frequency knob without orange dot in the center, large interval. 

Click on plus sign until 20 MHz. Once in 20 MHz, try clicking the plus button, it will respond.

Set the frequency value to 50 MHz, the value will then set to the max possible frequency which is 37.5 MHz.

Click on the plus button again, the frequency value will decrease to 20 MHz instead of staying at 37.5 MHz.

[image]

adisuciu, 2017-07-31:
Fixed with 9f2f0b1

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus