analogdevicesinc/scopy #208: Signal unstable at low frequencies

amiclaus, 2017-07-28:
Scopy version: 8406b30

I'm using a RC HPF (10k, 39pF) with the Network Analyzer- see https://github.com/analogdevicesinc/scopy/issues/168#issuecomment-314133868 for more info regarding HPF

I'm going on a 100Hz-10MHz sweep. At low frequencies the plot result is quite unstable.
Below is a comparison between M2K and Discovery using same setup.
[image]
[image]

damercer, 2017-07-28:
This could be due to the larger LSB in M2K ( 12bits vs 14 bits or 12 dB). Or, which gain range is the input stage set to? If it is in the +/- 25 V range the LSB is huge.

Doug

adisuciu, 2018-01-23:
This issue seems to be solved in the latest Scopy version. There are no visible spikes when the amplitude is low.
[image]
[image]

amiclaus, 2018-03-13:
On Scopy build: 2e31a75 (v0.11) the bug is fixed.

closed 2018-03-13 by amiclaus