analogdevicesinc/scopy #210: Logic Analyzer: Signal plot may be moved not in sync with the channel manager

drbautista, 2017-07-31:
OS: Windows 7 64bit
 Scopy Version:  fc64aff
 Firmware Version: v0.14

Open the logic analyzer instrument.

1. Click on the signal plot area.

[image]

2. The signal plots may be moved using the up and down arrow key from your keyboard but the channel manager won't move, leaving it behind.

[image]

*clicking on the channel manager area and clicking the up and down arrow key from the keyboard will move both the channels and signal plot in sync with each other.

AlexandraTrifan, 2017-07-31:
Fixed with commit 131e57e3883083403ac4618ebc60257a6ff72018.

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus