analogdevicesinc/scopy #211: Phase returns to 0 after selecting other channel

tagoylo, 2017-07-31:
Scopy build: [8406b304](https://github.com/analogdevicesinc/scopy/commit/8406b3043e72ebd5755094674c4a57aa839a565al)

Steps to reproduce:

Enable DIO0. Change phase to any value i.e. 45 deg. Then use the left/right arrows to select other channel or juct click to other channel.

Return back to DIO0, the phase that we set previously as 45 deg has now returned back to 0. But the display/trace stays the same, showing the shift. 

[image]

adisuciu, 2017-07-31:
Fixed with 47b6899

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

commit 47b68990 (2017-07-31): PatternGenerator - phase widget not updated when deselecting channel

Fixes #211
closed 2017-08-28 by amiclaus