analogdevicesinc/scopy #212: Pattern generator fails to generate clock frequency greater than 1MHz

tagoylo, 2017-08-01:
Scopy build: [9f2f0b17](https://github.com/analogdevicesinc/scopy/commit/9f2f0b17e3edbf50159bc79ead8c3b2f4fdff229)

Steps to reproduce:
Enable DIO0. Set as clock. Set clock frequency to 2MHz/5MHz/10MHz.

Monitor through oscilloscope. Enable built-in measurement feature to see frequency. The result is shown below when pattern generator frequency is set to 2MHz:

[image]

Also monitor through logic analyzer. Use cursors to know frequency. Use 200ns/div time base. The result is shown below when pattern generator frequency is set to 2MHz:

[image]

Duty cycle is also not consistent.

adisuciu, 2017-08-01:
@acostina  It looks like this is an HDL issue.

lclausen-adi, 2017-08-02:
What's the size of the buffer you are creating for this waveform? For good performance the buffer size should be at least 128 bytes. If it is smaller repeat it in software until the minimum is reached.

adisuciu, 2017-08-02:
For 10 MHz , the buffersize is 8. It looks like I forgot to update the "main" sampling frequency to 100MHz (it's still 80), but that doesn't seem to cause the issue. This is some code I've used to reproduce:
https://pastebin.com/raw/cbQjuZPa

lclausen-adi, 2017-08-17:
Increasing the minimum number of samples per buffer to 128 seems to avoid the issue.

rgetz, 2017-08-17:
Lars:

Should the kernel just reject buffers (fail and return error code) it knows are too small?

lclausen-adi, 2017-08-17:
The kernel doesn't know that the buffers are too small. This is a system design problem. It depends on the buffer size, the DDR memory performance and the data read rate. Smaller buffers will work fine with smaller samplerates or faster DDR speed.

rgetz, 2017-08-17:
Got it. Thanks

adisuciu, 2017-08-29:
Fixed with a050a81158054fee5616b1909f9dad5bc4a3ffa2

amiclaus, 2017-08-30:
On Scopy build: 5bb6466 (v0.7) the bug is fixed.

closed 2017-08-30 by amiclaus