analogdevicesinc/scopy #214: Cannot generate number pattern

tagoylo, 2017-08-02:
Scopy build: [9f2f0b17](https://github.com/analogdevicesinc/scopy/commit/9f2f0b17e3edbf50159bc79ead8c3b2f4fdff229)

steps to reproduce

Create a group with 3 channels. DIO0 to DIO2.
Change pattern to number pattern. Then input number, 3 for example.
In logic analyzer, the reading is always 1. Also checked it with discovery's logic analyzer and the result is the same.

[image]

adisuciu, 2017-08-02:
This happens due to no restriction to the buffersize. Buffersize must be a multiple of 4, but number pattern creates buffersize of 1. 

Fixed with commit 27e9deaa71c2397479d6a85ce7b349cac9d725f0

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus