analogdevicesinc/scopy #216: Choosing black as Edge, low and high color doesn't show black

tagoylo, 2017-08-03:
scopy build: [caf075e3](https://github.com/analogdevicesinc/scopy/commit/caf075e389be1320eba48722393f7a89eb949dd1)

Choosing black as Edge, low and high color doesn't show black. Instead the default colors are displayed. This happens on all channels. All other colors work except for black.

[image]

AlexandraTrifan, 2017-08-07:
I think we can replace this color anyway, since it is not visible on the current color configuration.

AlexandraTrifan, 2017-08-07:
The color was replaced with a color from the mockup.
Fixed with commit a5819c51e193e9a37fc4f3cada4a569de1ee70b1.

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus