analogdevicesinc/scopy #219: Logic Analyzer: Trigger activates on the second event

drbautista, 2017-08-07:
OS: Windows 7 64bit
 Scopy Version:  454babc
 Firmware Version: v0.14

*Open Digital IO instrument [This will serve as the trigger]
1. Set DIO0 as output and put it on Logic 0.
2. Run the Digital IO instrument

[image]

*Open the Pattern Generator instrument.
2. Set DIO1 to DIO8 to clock pattern and assign any frequency [any pattern will also work].
       -For this case use frequencies between 10Hz to 500Hz
3. Run the pattern generator

[image]

*Open Logic Analyzer instrument
3. Set time base to 5ms and position to 0ms
4. Set trigger to normal and Trigger Logic to OR
5. Set the trigger for DIO0 to rising edge
6. Click Single capture and see if the logic analyzer is waiting for a trigger.

[image]

7. Go back to the digital IO instrument and change the output of DIO0 from 0 to 1.
8. Go back to the Logic Analyzer instrument and notice that it didn't trigger the capture.
9. Try to change the DIO0's output from 1 to 0 then 0 to 1 again.
10. That's the time it will trigger a capture.

*Same behavior happens on falling edge and any edge trigger option on all the DIO channels.
*Same behavior also happens using the external trigger
*I also tried using different source for triggering, and the behavior is still the same for the logic analyzer

drbautista, 2017-08-08:
I have an additional observation regarding this bug.

The logic analyzer actually captures the signal on the first event but it will only show this captured signal on the second event of the trigger.

here's how I figured it out.

1. Set DIO9 as output and set to 0 on Digital IO instrument [this will be the trigger]
2. In the pattern generator instrument, group DIO0 to DIO7 and set as number pattern for this case I set a numerical value of 50. DIO8 will be a clock pattern at 100Hz

[image]

3. In the logic analyzer instrument, group DIO0 to DIO8, and set as parallel. Set DIO0:DIO0 . . . DIO7:DIO7 and DIO8 as CLK. Set rising edge trigger on DIO9.
4. Trigger the rising edge of DIO9 in the Digital IO instrument, and see that the logic analyzer is not showing any captured data.
5. Change in the pattern generator the numerical value, this case is 100. 

[image]

6. Trigger DIO9 again.

[image]

It shows the hex value of the Decimal 50 which is 32.

AlexandraTrifan, 2017-08-17:
Fixed with commit 0436400ae30b454b0b05e89c8d66e9d0290820e0.

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus