analogdevicesinc/scopy #220: Logic Analyzer: External trigger won't automatically turn off when activating any DIO channel's trigger within a group

drbautista, 2017-08-07:
OS: Windows 7 64bit
 Scopy Version:  454babc 
 Firmware Version: v0.14

Open Logic Analyzer instrument

1. Activate External Trigger from the Trigger settings
2. Group any DIO channels
3. Activate a trigger from any DIO channel within the group.
4. The external trigger won't automatically turn off.
[image]

*Activating a trigger of any DIO channel outside the group will automatically turn off the external trigger function.
*Turning on the external trigger function will automatically turn off the triggers within and outside DIO channel groups.

AlexandraTrifan, 2017-08-07:
Fixed with commit 44417160818dc9f94b056e7ff95f5ac80d6873c2.

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus