analogdevicesinc/scopy #221: Changing background color doesn't appear to be changed totally

tagoylo, 2017-08-08:
scopy build: [a5819c51](https://github.com/analogdevicesinc/scopy/commit/a5819c51e193e9a37fc4f3cada4a569de1ee70b1)

Changing background color doesn't appear to be changed totally. The selected color is transparent such that black is still visible and the selected background color doesn't appear as expected but I'm not totally sure how this is supposed to be implemented. Also in the new update regarding the black color, for the background color the only way to get back to black background is to reset instrument. 

[image]

AlexandraTrifan, 2017-08-08:
Commit b9089cf128873bd54a85b86d132e7ce801342651 introduces a fix for the issue with the black background color. Black is kept as an option for background colors, while another color is used as an option for 'edge', 'high' and 'low'.

amiclaus, 2017-10-10:
On Scopy build: 8f7e63f (v0.8) the bug is fixed.

closed 2017-10-10 by amiclaus