analogdevicesinc/scopy #223: Incorrect hexadecimal value in group channel

tagoylo, 2017-08-08:
Scopy build: [a5819c51](https://github.com/analogdevicesinc/scopy/commit/a5819c51e193e9a37fc4f3cada4a569de1ee70b1)

Steps to reproduce:

Create group with DIO1 to DIO4 in pattern generator.
Change pattern to binary counter. 
Monitor in logic analyzer. Create group with DIO1 to DIO4 use parallel as decoder.
Run instruments.
As shown below, the hexadecimal equivalent in logic analyzer doesn't correspond to what is seen in pattern generator.

The correct ouput only shows when selected channel is with DIO0 then follows the next channels i.e. DIO0, DIO1, DIO2 ... 

[image]

AlexandraTrifan, 2017-08-09:
The order in which the data lines are assigned is not important. The important thing is that they shouldn't have an index greater than the number of channels in the group. So the data lines should be assigned starting from D0 and no index should be skipped. 
Commit f388c022a311dcfa7d89cbf593604e0e9073463c was introduced to enforce this behaviour. The dropdowns are not available anymore. To change the data line assigned to a specific channel, the user can just reorder the channels inside the group.

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus