analogdevicesinc/scopy #224: Logic Analyzer: Trigger Logic AND not working

drbautista, 2017-08-09:
OS: Windows 7 64bit
 Scopy Version:  d6f5253
 Firmware Version: v0.14

1. Open Digital IO and set DIO0 and DIO1 as output. DIO0 = logic 0; DIO1 = logic 1 [this will serve as the trigger]
2. Open Pattern Generator and put clock signal on DIO2 and DIO3. [any type of signal will do]
3. Open Logic Analyzer and set DIO0's trigger to Low and DIO1's Trigger to High.
4. Set the trigger mode to normal and trigger logic to AND.

[image]

5. Run the Digital IO, Pattern Generator and Logic Analyzer instrument.
6. The Logic Analyzer won't execute a capture even if both triggers are satisfied with logic AND.

[image]

AlexandraTrifan, 2017-08-09:
The behaviour is the same when using the IIO-Oscilloscope. 
@acostina Can you take a look, please?

lclausen-adi, 2017-08-17:
Fixed with https://github.com/analogdevicesinc/hdl/commit/f9fe66694d7ff6489edadbf99f4947509e8b9c6a

Will be part of the next image update.

amiclaus, 2017-10-11:
On Scopy build: 8f7e63f (v0.8) the bug is fixed.

closed 2017-10-11 by amiclaus