analogdevicesinc/scopy #225: Logic Analyzer: Locked cursor mode not in sync with the signal plot

drbautista, 2017-08-10:
OS: Windows 7 64bit
 Scopy Version:  d6f5253 
Firmware Version: v0.14

1. Open Pattern Generator Instrument.
2. Enable DIO0 and set as pattern: Clock, Frequency: 100Hz and phase degrees: 50% [any setup will do] and run the instrument.
3. Open Logic Analyzer, enable DIO0 and set the time base to 5ms, enable cursor and measure the signal's frequency. The cursor's measured value is quite accurate.

[image]

4. Enable the lock function and try to move the cursor.

[image]

5. The cursor's reading is still the same, but the visual representation changed and not in sync with the signal plot.

AlexandraTrifan, 2017-08-21:
Fixed with commit 01dd5d00fad03eef00e77190d4a7f9b8d67b3a75.

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus