analogdevicesinc/scopy #226: Adding Lines/Points option for XY feature

amiclaus, 2017-08-11:
Currently, the graph for the XY feature is represented using only points:

[image]

Is it relevant to introduce the Lines/Points option for this feature? Similar to the ADI IIO Oscilloscope.

[image]

damercer, 2017-08-11:
Lines. Connecting the sample points should always default to lines.

Doug

dNechita, 2017-08-14:
With this commit https://github.com/analogdevicesinc/scopy/commit/bd7546d592fb2cce376f5c156150fe9ddd90c916 the XY plot defaults to displaying lines but there is also an option to switch to points.

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus