analogdevicesinc/scopy #227: Logic Analyzer: Scopy crashes when exporting csv file format

drbautista, 2017-08-17:
OS: Windows 7 64bit
 Scopy Version:  b241b20
 Firmware Version: v0.14

1. Open pattern generator and generate different signals on each channel. 
2. Open Logic Analyzer instrument and run both pattern generator and logic analyzer.
3. Click the export function and save as csv format.

[image]

4. After clicking save and confirm on the default option. Scopy will crash.

[image]

AlexandraTrifan, 2017-08-29:
Fixed with commit 9487204b3ec2d8a84336f58795565d2fd4d05965.

amiclaus, 2017-09-05:
On Scopy build: 49a4f93 (v0.7) the bug is fixed.

closed 2017-09-05 by amiclaus