analogdevicesinc/scopy #228: After a run on network analyzer, changing the min/max frequency distorts the plot

tagoylo, 2017-08-17:
Scopy build: [9bf99972](https://github.com/analogdevicesinc/scopy/commit/9bf99972f4be2691410a1e1d6d2ddfc0066482de)

Steps to reproduce:

 Using RC LPF circuit fc at ~530 Hz
Sweep range 10 Hz to 1MHz

After a run on network analyzer, click on the ± button or input new max frequency. The plot is now distorted as seen in the picture below. Also clicking  ± button will cause a change of 2 intervals. Sometimes, after few clicks, scopy will then crash. 

[image]

DanielGuramulta, 2017-08-21:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/f8be02839d5b65ede70b05712ef380a3d35788b4

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus