analogdevicesinc/scopy #234: Adding of cursor/position option

tagoylo, 2017-08-23:
Scopy build: [018987a2](https://github.com/analogdevicesinc/scopy/commit/018987a29e4cfb9b893146feaf5d9b3ef08619f4)

Adding of cursor/position option to see exact frequency value at -3dB point, also to track phase shift along with the magnitude. This will provide convenience for the user in using the network analyzer. Instead of dragging/moving the mouse cursor along the trace to find the -3dB point of a LPF frequency response, for example. The user will just go to the cursor/position option to locate the point.

adisuciu, 2018-03-07:
Implemented with https://github.com/analogdevicesinc/scopy/pull/372

amiclaus, 2018-03-12:
On Scopy build: 6bd7075 (v0.11) the feature is implemented.

closed 2018-03-12 by amiclaus