analogdevicesinc/scopy #235: Logic Analyzer: Toggling the Trigger Mode without Stopping the instrument makes the triggering unstable

drbautista, 2017-08-24:
OS: Windows 7 64bit
 Scopy Version:  d0f3ca5
 Firmware Version: v0.14

A. From Normal Mode to Auto Mode
      1. Open Digital IO instrument and set DIO0 to output and logic 0.
      2. Open Pattern Generator, Disable Digital DIO0 and set the rest of channels to any signal.
      3. Open Logic Analyzer, set DIO0's trigger to rising edge toggle the trigger mode to normal.
      4. Run the three instruments. Try to trigger DIO0 on normal mode to see if the logic analyzer is working.
      5. Toggle the Trigger mode to Auto. Auto mode should automatically capture signals even when there's no trigger event, but this time it won't capture data.
[image]
      6. It will continuously start capturing data by activating a trigger once.

B. From Auto Mode to Normal Mode
      1. Open Digital IO instrument and set DIO0 to output and logic 0.
      2. Open Pattern Generator, Disable Digital DIO0 and set the rest of channels to any signal.
      3. Open Logic Analyzer, set DIO0's trigger to rising edge, toggle the trigger mode to auto.
      4. Run the three instruments, Logic analyzer should start capturing in auto mode. 
      5. Toggle the trigger mode to Normal mode. the capturing should stop because DIO0's trigger is set to rising edge, but the Logic Analyzer continuously captures data.
[image]
      6. The only way to make the instrument start waiting for the trigger again is by changing the trigger option of DIO0 or by toggling to auto mode, then stop the instrument and toggle to back to normal mode before starting the instrument again.

AlexandraTrifan, 2017-08-24:
Fixed with commit 55c6e7cafde50426d866f6ae7fb7ea70caf015a6.

amiclaus, 2017-08-28:
On Scopy build: be46948(v0.7) the bug is fixed.

closed 2017-08-28 by amiclaus