analogdevicesinc/scopy #236: Incorrect Initial channel state

amiclaus, 2017-08-29:
Scopy version: be46948

Delete Scopy.ini file. Open DigitalIO Instrument. The channel state is set to low while all channels are set as input.
[image]

By changing the type of the channel(in/out) or run the instrument, all the channels are refreshed and set to high.
[image]

Since all the channels are set as input by default, the high state should appear as default.

DanielGuramulta, 2017-08-29:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/d4e3291f04476142a971f9e3e3b6ea394fafffd5

amiclaus, 2017-08-30:
On Scopy build: 5bb6466 (v0.7) the bug is fixed.

closed 2017-08-30 by amiclaus