analogdevicesinc/scopy #237: Incorrect RMS value

amiclaus, 2017-08-29:
Scopy version: be46948

Generate via external/Scopy a square wave with frequency 40kHz, amplitude 5V.

Oscilloscope RMS value of the signal:
[image]
Voltmeter RMS value of the signal:
[image]
It seems that the voltmeter displays the RMS with an error of ~0.3V (should display around 2.5V)

AlexandraTrifan, 2017-08-31:
Fixed with commit 8651b8326c5ba764ef613c85e50add9d61a36e20.

amiclaus, 2017-09-01:
On scopy build 40e7d80 (v0.7) the bug is fixed.

tagoylo, 2018-10-08:
This issue has reappeared in V1.0.2.
Scopy build: [1d173e63](https://github.com/analogdevicesinc/scopy/commit/1d173e633165f0632036963f0b15180598c5add9)

[image]

DanielGuramulta, 2018-10-10:
What hardware revision do you have?

tagoylo, 2018-10-10:
I'm using a Rev c board.

ReiRue, 2018-10-23:
Keep in mind what you are doing:
A square wave has an unlimited number of harmonics!
In your example:

frequenzy  relative amplitude
 40kHz        1
120kHz        1/3
200kHz        1/5
280kHz        1/7
...           ...
(check it with the integrated spectrum analyser)

Your voltmetet has an "integrated" low pass filter with an
corner frequency of about 40kHz
-> this filter will attenuate all harmonics above 40kHz!
-> results in a reading below 2.5V

My interpretation:
  There is no bug!

amiclaus, 2019-02-27:
fixed on Scopy build: 91c4f0c

closed 2019-02-27 by amiclaus