analogdevicesinc/scopy #242: Oscilloscope: Second Math channel measurements disappear when the first channel is removed

drbautista, 2017-08-31:
OS: Windows 7 64bit
 Scopy Version:  5bbb6466
 Firmware Version: v0.14

Steps to reproduce:

1. Input any signal on the oscilloscope channels.
2. Add at least two Math channels and input any function.
3. Delete the channel located beside channel 2

[image]

4. The next channel's measurement data will not be visible.
[image]

dNechita, 2017-08-31:
Fixed with: https://github.com/analogdevicesinc/scopy/commit/174536996b70d92e3e9c1fa273d17f83d905e7b6

amiclaus, 2017-09-01:
On scopy build 40e7d80 (v0.7) the bug is fixed.

closed 2017-09-01 by amiclaus