analogdevicesinc/scopy #243: Use Trace Averaging for XY plot

amiclaus, 2017-08-31:
Would it be an idea to use trace averaging for the XY plots?
Maybe we can obtain nicer plots than, for example, this one: 
[image]
(photo took from ADALM2000 Lab Material - [Activity 8. Stabilized current source (BJT)](https://wiki.analog.com/university/courses/electronics/electronics-lab-8))

rgetz, 2017-08-31:
Do you want averaging? Or a "best fit" curve, with the actual math equation?

Best fit would seem to be to be more valuable.

- Robin

amiclaus, 2017-09-01:
There are several types of curve fitting based on different equations.
For different plot shapes, there might be different curve fitting equations that will construct the plot properly. 
Therefore, should we add multiple fitting methods and will be the user's choice which one to apply? (Like here : https://www.mycurvefit.com/)

DanielGuramulta, 2017-09-21:
I've implemented a filter: https://github.com/analogdevicesinc/scopy/commit/9dc63dd94ba8d433b73d1696722f2a9079ca34d3 that will create an average curve to be drawn on the plot.

closed 2024-09-24 by adisuciu