analogdevicesinc/scopy #244: Scopy crashes when connecting to M2K on Linux

amiclaus, 2017-08-31:
Follow the build guide available here: https://github.com/analogdevicesinc/scopy/wiki/Building-Scopy-on-Linux

Run Scopy, Connect to M2K. Scopy crashes.

Backtrace attached: 
[backtrace-issue.txt](https://github.com/analogdevicesinc/scopy/files/1267626/backtrace-issue.txt)

AlexandraTrifan, 2017-09-04:
I can reproduce this issue using an older version of GNU Radio (the last commit was on 14 Aug). Updating the GNU Radio library fixes this issue. 
I think that, from time to time (or when a new release should be tested), all the dependencies should be checked to make sure we are using the latest version.

amiclaus, 2017-09-05:
On Scopy build: 49a4f93 (v0.7) the bug is fixed.

closed 2017-09-05 by amiclaus