analogdevicesinc/scopy #245: Generated clock frequency is not accurate

tagoylo, 2017-09-04:
scopy build: [40e7d80c](https://github.com/analogdevicesinc/scopy/commit/40e7d80cb9767c8bcaa8d63957d34c9345addc26)

When specifying clock frequency, other chosen frequency has inaccurate output as monitored by the oscilloscope.

Steps to reproduce:
open pattern generator
enable DIO0 and set clock frequency to: 170 kHz, 240 kHz, 290 kHz, 470 kHz, 27 MHz, 34-49 MHz, etc.
set duty cycle to 50 %
Monitor in oscilloscope.
The frequency read is not as what is set.

[image]

adisuciu, 2017-09-04:
Fixed with fb5a0f8

amiclaus, 2017-09-05:
On Scopy build: 49a4f93 (v0.7) the bug is fixed.

closed 2017-09-05 by amiclaus