analogdevicesinc/scopy #247: Time alignment of Analog and Digital outputs?

damercer, 2017-09-04:
If you set one of the analog outputs to generate a 10 KHz sinewave in the signal generator and connect it to one scope input and set one of the digital outputs to generate a clock signal at 10 KHz in the pattern generator and connect it to the second scope input. With the scope triggered on the sine wave, observe the relative time alignment of the two waveforms. It will slowly drift over time, seconds to minutes. The higher the output frequency the faster the apparent drift. My original question was going to be is it possible to align the analog and digital outputs in time in a deterministic way but, because of this drift, obviously not.

My guess it that this drift is caused by the fact that the analog output DAC sample clock is 75 MHz and the digital output clock is 100 MHz which are off by a small fraction of a Hz. The solution would be to have the DAC sample clock, the ADC sample clock and the digital sample clock all be the same 100 MHz master clock.

This will likely need to be changed / fixed in the firmware and not the GUI.

Thanks

Doug

acostina, 2017-09-08:
At this point, the pattern generator clock is generated by the FPGA. There is the option for the clock to be driven by data[0] pin instead. 
In AD9963, the ADC path is driven directly by the reference clock and the DAC path is driven by a derived clock, which is generated by the DLL. Driving the DAC side at 150MSPS with interpolation has some benefits on the quality of the generated signal, that's why we choose to use it instead of 100MSPS.
In issue 248, if we want, when reconfiguring the DAC to work at 100 MSPS, we can take into consideration disabling the DLL and using the reference clock directly to drive the DAC. 
We do not have access to the reference clock in the FPGA, only as the digital interface clock from the ADC or DAC. It's possible to use one of those clocks, but this will require changes in the architecture of the HDL and software drivers.

damercer, 2017-09-08:
This is not necessarily a show stopper at this point but my guess is that at some point an advanced user ( mixed signal ) will notice the problem and ask for a solution.

Doug

closed 2024-09-24 by adisuciu