analogdevicesinc/scopy #253: After a math channel is deleted, another channel might be selected

DanielGuramulta, 2017-09-07:
Add some math channels, select a channel and delete another one:
[image]
In this case, Math3 is the selected one, but after we delete Math1:
[image]
Math2 will be the now selected channel.
Another channel should be selected only when deleting the current selected channel.

DanielGuramulta, 2017-09-07:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/30cf24d0acf636c47a7d6bc0cbd13d8d3357dba6

amiclaus, 2017-09-08:
On Scopy build: 152efe0 the bug is fixed.

closed 2017-09-08 by amiclaus