analogdevicesinc/scopy #255: Duty cycle is inaccurate at high frequencies, >1MHz

tagoylo, 2017-09-11:
Scopy build: [152efe0c](https://github.com/analogdevicesinc/scopy/commit/152efe0cd986e08bfb7b4e32c36ba18284214d3e)

When clock frequency is set to high, i.e. 10 MHz and change the duty cycle to value other than 50%, the result monitored in the oscilloscope is far from what is set.

[image]
[image]

damercer, 2017-09-11:
At 10 MHz output with a 100 MHz sample rate you only have 10 samples per cycle or 10% steps in the duty cycle. So how can you be allowed set the duty cycle with more resolution than that?

Doug

adisuciu, 2017-09-20:
Fixed with 1e4e8957717b242a745b33f670dd6b18cf51f321

amiclaus, 2017-10-10:
On Scopy build: 8f7e63f (v0.8) the bug is fixed.

closed 2017-10-10 by amiclaus