analogdevicesinc/scopy #258: Reset instrument doesn't reset plot to default  state

tagoylo, 2017-09-12:
Scopy build: [152efe0c](https://github.com/analogdevicesinc/scopy/commit/152efe0cd986e08bfb7b4e32c36ba18284214d3e)

Steps to reproduce:

Open pattern generator and enable DIO0, set clock frequency as is, then click Run.
Open logic analyzer and enable DIO0, click Run.
Stop first logic analyzer then the pattern generator.
Click Reset Instrument of logic analyzer.
The plot doesn't reset to its default state.

[image]

AlexandraTrifan, 2017-09-13:
Fixed with commit b1162156655111557fa0fc112ab89eac4fafa9de.

amiclaus, 2017-10-10:
On Scopy build: 8f7e63f (v0.8) the bug is fixed.

closed 2017-10-10 by amiclaus