analogdevicesinc/scopy #259: Closing a detached instrument disrupts main window

tagoylo, 2017-09-13:
Scopy build: [0700bee3](https://github.com/analogdevicesinc/scopy/commit/0700bee34e5db9f4840ae45f2b32b66064725bca)

Detach an instrument, ex. Digital IO
Then close it
Select again that instrument in the instrument menu in main window
After selecting, a blank window, appears. See screenshot below

[image]

Then if you close the blank window the main window will appear as seen below.

[image]

DanielGuramulta, 2017-09-13:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/9332e85355ab4859a9e97306ec54fe484ce5842f

amiclaus, 2017-10-10:
On Scopy build: 8f7e63f (v0.8) the bug is fixed.

closed 2017-10-10 by amiclaus