analogdevicesinc/scopy #262: Horizontal Trigger/Channel Cursors Handles dissapear

amiclaus, 2017-09-18:
Scopy version: 316a80e

Open Scopy in minimum size window, open Oscilloscope, enable Display All Measurements. Move the Vertical Trigger cursor to the far left/right of the plot:

[image]

Restart Scopy in minimum size window. The Horizontal Trigger Cursor Handle dissapears along with the channel position handlers:
[image]

DanielGuramulta, 2017-09-25:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/6e6316fe6f409a793b7cdc4b3d13835475c25dec

amiclaus, 2017-10-10:
On Scopy build: 8f7e63f (v0.8) the bug is fixed.

closed 2017-10-10 by amiclaus