analogdevicesinc/scopy #265: Changing values/measurement unit improvement

amiclaus, 2017-09-18:
Open i.e. Signal Generator Channel Settings. Suppose that the set value is 5V.

[image]

Now I want to change the value to 200 mV. As we do on paper, first we write the value and then the measurement unit. After changing the value to 200, when trying to change the measurement unit to mV, due to defocus, the value changes to automatically to 5 (maximum accepted value in Volts).

An improvement would be in this situation to automatically change the measurement unit to a smaller one (i.e. mV) if the set value is greater than the accepted one.

Not sure if it is bug, so I will consider it an enhancement

septimiusarca, 2018-02-21:
Fix with commit [75170b6](https://github.com/analogdevicesinc/scopy/pull/374/commits/75170b6a44d6c97a53d213c250d369ead637db9b).

amiclaus, 2018-03-12:
On Scopy build: 6bd7075 (v0.11) the feature is implemented.

closed 2018-03-12 by amiclaus