analogdevicesinc/scopy #269: Duty cycle gradually becomes 0 as frequency is increased

tagoylo, 2017-09-21:
Scopy build: [1e4e8957](https://github.com/analogdevicesinc/scopy/commit/1e4e8957717b242a745b33f670dd6b18cf51f321)

Steps to reproduce:

Enable DIO0
Set clock frequency to 1MHz.
Enable fine interval, +-1 unit interval, orange dot in center.
Then max out duty cycle to 100%.
Click + to increase frequency.
Monitor Duty cycle value. It decreases until 0. And as seen in the screenshot below, the wheel doesn't correspond to the value.

[image]

adisuciu, 2017-10-09:
Fixed with  bad81d4

amiclaus, 2017-10-10:
On Scopy build: 8f7e63f (v0.8) the bug is fixed.

closed 2017-10-10 by amiclaus