analogdevicesinc/scopy #271: Can't increase duty cycle value, it is stuck

tagoylo, 2017-09-21:
Scopy Build: 

Steps to reproduce

Enable DIO0
Set duty cycle to 0.
Set clock frequency to any of the following:
- 3 MHz	
- 6-7 MHz
- 9 MHz 
- 11-15 MHz				
- 18-19 MHz					
- 21-24 MHz					
- 26-30 MHz			
- 33-34 MHz
- 36 MHz
- 38 MHz
- 42 MHz
- 44 MHz
- 46-49 MHz

For example, 3 MHz.		
Then click + to increase duty cycle.
The value is stuck at 3. It does not increase further. The same as with the other mentioned frequency.

[image]

adisuciu, 2017-09-25:
Fixed with cdeab7f07118c2367adbc8c9e65a87a2f6617b98

amiclaus, 2017-10-11:
From my point of view, on 8f7e63f (v0.8), this bug partially fixed: if using the fine mode in order to increase the duty cycle, the value remains stuck. This happens due to the small nr. of samples available on each period of the signal and implicitly the duty cycle values available is reduced. Therefore, from the design point of view, something should be made: either make the fine mode unavailable in this kind of situation, either coarse mode and fine mode have the same behavior.

amiclaus, 2017-11-21:
On Scopy build: c8ad583 (v0.9) the bug is fixed.

closed 2017-11-21 by amiclaus