analogdevicesinc/scopy #272: Channel positioned above range on plot

amiclaus, 2017-09-22:
Scopy version: d3a3e61

Set Volts/Div to 10V. When moving the channel position above 25V, the position in the channel setting remains at 25V. Since the range is defined +-25V and there are 10 divisions on vertical for the plot, the Volts/Div value should be limited to +-5V.
[image]

dNechita, 2017-09-26:
I think this issue needs debating.
One approach of handling this is the above mentioned solution where we limit the plot vertical axis to +- 25V. Disadvantages:
 - M2K can measure a little bit more than +-25V. I think the actual interval is around +-30V. But the plot won't allow anything beyond 25V.

The other option is to keep things as they are but make sure widgets (offset level, trigger level) don't go beyond 25V. Disadvantages:
 - Creates confusion because the widgets (offset and trigger handles) won't go any higher that the 25V threshold
 - The signal might get clipped due to ADC saturation even though it didn't reach the top (or bottom) plot limit just because the plot interval is larger that what the hardware can measure

DanielGuramulta, 2017-10-02:
Fixed with commit: https://github.com/analogdevicesinc/scopy/commit/b04cedc55358e54e563e0e510fd67fefff3f98f3

amiclaus, 2017-10-10:
On Scopy build: 8f7e63f (v0.8) the bug is fixed.

closed 2017-10-10 by amiclaus